cnlohr/ch32fun #560: [minichlink][esp32s2-programmer] Error: Got error -1 when getting hid feature report. (Size 255/255)

Plaque-fcc, 2025-03-12:
This is what I get when `make flash` in examples (specifically, `examples/blink_raw/`):

```
../../ch32fun//../minichlink/minichlink -w blink_raw.bin flash -b
Error: Got error -1 when getting hid feature report. (Size 255/255)
Error: Got error -1 when sending hid feature report.
make: *** [../../ch32fun/ch32fun.mk:268: cv_flash] Error 247
```

Tested with two identical ESP32 S2 Mini boards, I utilized one of them as a programmer, and this `minichlink/minichlink` tool worked fine with it.

I'm trying to identify how to reproduce this issue stably.

Plaque-fcc, 2025-03-12:
Bisecting the history of changes, I managed to identify the change which breaks support of ESP32S2-based programmers, seems like:
297d9f86afbabdd1ab3e0d2c19db4d6b840b8f03

cnlohr, 2025-03-13:
I believe there was a breaking change before/after that.  I now am reporting the version number of the ESP32-S2 back to the host, so it is possible to version things.  But yes, you would need to update the ESP32-S2 at this time :(.

If you're up for doing a PR, you could potentially do the old code for if there is no version number.

cnlohr, 2025-03-13:
I would personally be interested in supporting older programmers for the majority of users.  My changes were mostly surrounding making sure that we could program a broader swath of boards moving forward.

cnlohr, 2025-03-14:
Please do look into it.  If you want, there will be many people available on a Discord call tomorrow at 2PM US Eastern Time if you want to join, hit me up on Discord. @Plaque-fcc

Plaque-fcc, 2025-03-14:
I did full flash-erase on those ESP32 S2 chips.
I assembled the `esp32s2-cookbook/ch32v003programmer` firmware again from the top of `master` branch.
I double-checked it all. It is very much likely that re-assembling the programmer firmware helped, because I can't reproduce the error anymore.

Finally.