cnlohr/ch32fun #565: CH59x printf does weird stuff on strings longer than 7 chars

biemster, 2025-03-15:
Apparently I did not test this well enough, this needs fixing. Just putcharring with a delay works, so probably printf does it too quick?

TommyMurphyTM1234, 2025-03-15:
Perhaps it needs some sort of handshake/checking like the non-CH59x case? 

- https://github.com/cnlohr/ch32fun/blob/f1b8b67120886d47edbb2ad26ba6696a4985ae3d/ch32fun/ch32fun.c#L1319-L1329

biemster, 2025-03-15:
> Perhaps it needs some sort of "ready for transmit" handshake/checking like the non-CH59x case?

that is very possible, I just copied this from the EVT. Very possible that I missed something like this, thanks!

cnlohr, 2025-03-17:
[image]

cnlohr, 2025-03-17:
[image]

biemster, 2025-03-21:
this was fixed in #570 (i thought that gh would close this automatically after merge)

closed 2025-03-21 by biemster