cnlohr/ch32fun #572: Perhaps faulty declaration of GPIO_Speed_50MHz in file "ch32v003hw.h" by using a ch32v003

jjflash65, 2025-03-21:
At first: please excuse my bad english due to the fact, that I'm a (very) old (60+ years) native speaken german. I have learned english 50 years ago at school and this in not sufficient quality.

So, I'm very new to ch32 microcontrollers (most used chips are STM32, STM8, AVR and PFS154), but it's never too late to learn new things and so I'm studying your codes by reading the reference manual at the same time.

The goal is to adapt ch32fun to my way of programming.

By doing this, I've found a discrepancy between your declaration of your enumerator in ch32v003hw.h in lines 3487 to 3493. In this enumerator you declare at the third element "GPIO_Speed_50MHz" but in the ch32v003 referene manual version 1.7 is written, that 2 Bits of register GPIOx_CFGLR defines the outputmode. Here is writtenwor

11: Output mode, maximum speed 30MHz.

For a fast fix I've added

#define GPIO_Speed_30MHz    3

But I'm not sure what is right.

-----------------------------------------------------------

An off-topic comment to such a lot of work from you:

To learn how to program a ch32v003 controller I've ordered two times a wch-linkE programmer but both times when the flash-programmers delivered to me, I've became a wc-link programmer (without "E"). So I'm using now ardulink to do the flash job by hoping the third delivery of a programmer is the right one. 

But: Your Makefiles in the example doesn't support ardulink although the host program minichlink can do the job. Perhaps you want to change your makefile's in the examples (for myself I have already done that).

------------------------------------------------------------

At the End: nice regards and thousand thanks for doing the job with ch32fun

cnlohr, 2025-03-21:
We overload both in the `GPIO_CFGLR_PIN_MODE_Typedef` enum

```
...
	GPIO_CFGLR_OUT_50Mhz_PP = 3,
	GPIO_CFGLR_OUT_30Mhz_PP = 3,
...
```

When I looked at the output for twiddling the output, it looks more like I would call a 50MHz output, and originally the reference manual called it 50MHz, but later both the reference manual and Moun River revised the value to 30MHz.  We dragged our feet, but now, code written with either 50 or 30 MHz should compile fine. 

Your use of the `GPIOSpeed_TypeDef` is an oversight, I will change it to overload that as well.  Thank you for catching it.

cnlohr, 2025-03-21:
I didn't go through the normal PR path if anyone sees this, just committed to `master` since it was simple.

cnlohr, 2025-03-21:
@jjflash65 if you don't mind opening a new issue about the ardulink comment that would be helpful. I have no experience with that, but the folks who work on that may see it and pick it up.

commit e988a3c4 (2025-03-21): Alias 50 and 30MHz speeds in GPIOSpeed_TypeDef

Closes #572
closed 2025-03-21 by cnlohr