cnlohr/ch32fun #583: strchr - Bug or Feature ?

jjflash65, 2025-04-07:
In first - like ever - please excuse my bad english, it's neither good nor nice cause my native language isn't english.

I'm relatively new to CH32V003 and I'm evaluating possibilities of this chip. So I wanted to port some of my sources from STM32 and STM8 to CH32.

To my great surprise, the relatively simple source codes did not work. Namely those codes who are using strchr which are declared in <string.h>

I have tested the following program on every platform I use:

`
  #include <string.h>

  int main(void)
  {
    char string[15];
    char *ptr, c = 'r';

    smallio_init();

    strcpy(string, "This is a string");
    ptr = strchr(string, c);
    if (ptr)
      printf("The character %c is at position: %d\n\r", c, ptr-string);
    else
      printf("The character was not found\n\r");

    while(1);
  }`

Every time the output was:

`The character was not found`

With CH32FUN the output is:

`The character z is at position: 16`

An original string.h discription says:

<b>char *strchr(const char *string, int c);</b>

The strchr() function finds the first occurrence of a character in a string. The character c can be the null character (\0); the ending null character of string is included in the search.

The strchr() function operates on null-ended strings. The string arguments to the function should contain a null character (\0) that marks the end of the string.

<b>Return Value</b>
The strchr() function returns a pointer to the first occurrence of c that is converted to a character in string. The function returns NULL if the specified character is not found.

In ch32fun.c
https://github.com/cnlohr/ch32fun/blob/master/ch32fun/ch32fun.c#L384

I found the source of strchr. This code finds the first occurrence of c, but if the searched char isn't present, the function returns pointed to the zero-byte at the end of the string (and not NULL).

For me I had replaced the function with this:

<pre>
/* ----------------------------------------------------------
                           strch
                           
     The strchr() function returns a pointer to the first 
     occurrence of ch that is converted to a character in
     string. The function returns NULL if the specified 
     character is not found.
   ---------------------------------------------------------- */

char *strch(const char *str, char ch)
{
  while(*str)
  {
    if (*str == ch) { return (char *)str; }
    str++;
  }
  if ((*str == 0) && (ch== '\0')) return (char *)str;
  return NULL;
}</pre>

Maybe you want to replace it too ?

TommyMurphyTM1234, 2025-04-07:
> ```
> char string[15];
> ...
> strcpy(string, "This is a string");
> ```

The code above is problematic as it involves a buffer overrun. You're copying 17 bytes (16 character string plus `'\0'` `NUL` terminator) to a 15 byte buffer!

jjflash65, 2025-04-07:
I'm sorry, you are right, but this was not the problem. In my first try
the string was shorter. The problem with not NULL return of the function
strchr have nothing to do with the length of array.

If you change

char string[15]

to

char string[100]

you will the, that the origin code doesn't do that, what he should do

Nice greetings from germany,

Ralph

TommyMurphyTM1234, 2025-04-07:
Sorry, I don't really understand your previous post. It would help to provide a test case that illustrates the issue. You're also referring to an alleged issue with `strchr()` but citing code implementing `strch()`. The ch32fun implementation of `strchr()` is here and looks fine to me: 

- https://github.com/cnlohr/ch32fun/blob/f61d201e95c7c0508c5860142da99c1e9b0f7975/ch32fun/ch32fun.c#L384-L390

jjflash65, 2025-04-07:
The only error that I have made is to define a to short array in my test case. At the moment I have enhanced my test case to a array with 100 elements !!!

If you run the code on every other plattform, the output will be:

The character was not found

If you run strchr with the code you linked (from ch32fun.c), the output will be:

The character z is at position: 16

The error is: the code in strchr from ch32fun will never return a NULL - Pointer so it doesn't work fine for me.
Try it!

TommyMurphyTM1234, 2025-04-07:
You need to provide a small self contained reproducible test case. Your original code is erroneous. It also calls `smallio_init()` whose implementation is not provided so we have no idea what it does. For all we know it could corrupt something causing subsequent problems. 

I can't see any problem with the ch32fun implementation of `strchr()` which I quoted and linked to above and which isn't the same as the implementation (of `strch()`?) that you mention in your original post.

jjflash65, 2025-04-08:
So, I have made a test case for ch32v003 with a function call strchr() out of ch32fun.c and implemented own function renamed to my_strchr() to run both function in one program. Try this, and you will see the difference:

<pre>
// strchr_error.c

// shows an error in original CH32FUN framework

#include "ch32fun.h"
#include <stdio.h>
#include <string.h>

/* ----------------------------------------------------------
                           my_strchr

     The strchr() function returns a pointer to the first
     occurrence of ch that is converted to a character in
     string. The function returns NULL if the specified
     character is not found.
   ---------------------------------------------------------- */
char *my_strchr(const char *str, char ch)
{
  while(*str)
  {
    if (*str == ch) { return (char *)str; }
    str++;
  }
  if ((*str == 0) && (ch== '\0')) return (char *)str;
  return NULL;
}

/* ----------------------------------------------------------
                            main
   ---------------------------------------------------------- */
int main(void)
{
  char string[100];
  char *ptr;
  char c;

  SystemInit();

  c= 'z';

  strcpy(string, "This is a string");
  printf("\n\r---------------");
  printf("\n\rsearching the char ' %c ' in string ' %s '",c,string);
  ptr = strchr(string, c);
  printf("\n\r---------------\n\r");
  printf("\n\rResult using strchr() out of CH32FUN\n\r");
  if (ptr)
    printf("\n\rThe character %c is at position: %d\n\r", c, ptr-string);
  else
    printf("\n\rThe character was not found\n\r");
  printf("\n\r---------------");

  ptr = my_strchr(string, c);                       // renamed function to use both version of strchr
  printf("\n\rResult using my_strchr()\n\r");
  if (ptr)
    printf("\n\rThe character %c is at position: %d\n\r", c, ptr-string);
  else
    printf("\n\rThe character was not found\n\r");
  printf("\n\r---------------\n\r");

  while(1);
}

</pre>

And the output of this test case is:

<pre>
---------------
searching the char ' z ' in string ' This is a string '
---------------

Result using strchr() out of CH32FUN

The character z is at position: 16

---------------
Result using my_strchr()

The character was not found

---------------
</pre>

cnlohr, 2025-04-15:
Hmm, indeed, this does look like it may be a bug. 

This would be correct behavior for strchrnul, but not strchr

>        The strchrnul() function returns a pointer to the matched
>       character, or a pointer to the null byte at the end of s (i.e.,
>       s+strlen(s)) if the character is not found.

Is a possible avenue forward:

```c
WEAK char *strchr(const char *s, int c)
{
	c = (unsigned char)c;
	if (!c) return (char *)s + strlen(s);
	for (; *(unsigned char *)s != c; s++) if (!*s) return 0;
	return (char *)s;
}
```

cnlohr, 2025-04-15:
@jjflash65 Can you test with the following:

```
WEAK char *strchr(const char *s, int c)
{
	c = (unsigned char)c;
	do
	{
		char ts = *s;
		if( ts == c ) return (char*)s;
		if( !ts ) return 0;
		s++;
	} while(1);
}
```

I think this may be the optimal implementation?

jjflash65, 2025-04-15:
This implementation of strchr works fine for me. :-) thanks a lot for fixing !

closed 2025-04-15 by jjflash65

cnlohr, 2025-04-15:
@jjflash65 can you confirm that this PR fixes it? https://github.com/cnlohr/ch32fun/pull/587