cnlohr/ch32fun #592: Cannot get CH32V006 UART

VynDragon, 2025-04-26:
Hello,

I cannot get CH32V006 to output uart.

My setup is a breakout board with CH32V006F8P6, 3v3 to VDD and GND to VSS, UART on PD5/6.

This setup works on mounriver for uart.

I have set CH32V006 to 1 in funconfig, and makefile to target ch32v006.

Using uartdemo, the led blinks as expected, but nothing comes out of PD5

dwillmore, 2025-04-26:
I just compiled it for 003 and 006 and tested it on similar boards and they both look the same.  Pin 2?

dwillmore, 2025-04-26:
You set the target in the Makefile:
```
all : flash

TARGET:=uartdemo

TARGET_MCU?=CH32V003
include ../../ch32fun/ch32fun.mk

flash : cv_flash
clean : cv_clean
```

dwillmore, 2025-04-26:
Simply change the TARGET_MCU line to the model you're building for.  You'll probably want to do a 'make clean' just to be safe.

VynDragon, 2025-04-26:
Yea, I did that, that did not work.

VynDragon, 2025-04-26:
Ok i see what I did wrong, i set 
```#define CH32V006           1
```
in funconfig.h. Thanks for your help.

closed 2025-04-26 by VynDragon

cnlohr, 2025-04-29:
For anyone else wondering, this is no longer how you should do it.  You should update it in the Makefile so it can be aware of the 006's flash / RAM