cnlohr/ch32fun #594: CH5xx consolidation

biemster, 2025-04-30:
There is still a bit of work on the ch5xx series to be done to have a proper foundation for when the ch570/2 hits general availability, after which I suspect most attention will go to those.
I'll work through the following list defaulting everything to ch570 except a different mcu is specified:
- [ ] First ofc get the ch57x PR ready (#561)
- [ ] Consolidate the ch5[789]x demos into ch5xx_examples. First the `blink` and `debugprintfdemo`, after that every separate example needs testing on all 4 series (ch59x, ch582, ch573, ch570 (these are the boards I have))
- [ ] `minichlink` support for all of them (@monte-monte is doing awesome work here currently)
- [ ] Copy the complete SFR's of these series to their respective `_hw.h` headers, so all peripherals are accessible (this is an SDK after all)
- [ ] BLE/RF examples

cnlohr, 2025-04-30:
With the change to #ifdef the CH570, and changing the address of DMDATA0, blink and printfdemo both work on the CH570.

You can test however you'd like by using the app in ram example.

I think you should be able to consolidate the _hw.h's some and #ifdef within the _hw.h between the different chips in the ch5xx series.

biemster, 2025-05-02:
> You can test however you'd like by using the app in ram example.

`chprog.py` from https://github.com/wagiminator/MCU-Flash-Tools supports the 570 over ISP now as well, although this of course does require reconnecting all the time which is really annoying. Maybe we should make a `run_in_ram` example for the ch570 while @monte-monte is finalizing the minichlink support?

cnlohr, 2025-05-03:
I added a run_in_ram example that works with the ch570 in the examples folder, called https://github.com/cnlohr/ch32fun/tree/master/examples/app_without_flash

biemster, 2025-05-04:
> I added a run_in_ram example that works with the ch570 in the examples folder, called https://github.com/cnlohr/ch32fun/tree/master/examples/app_without_flash

How would I go about supporting the .highcode section in the BLE and RF lib using this example? Just add `*(.highcode)` and `*(.highcode.*)` in the `.text :` section in the linker script will suffice?

monte-monte, 2025-05-05:
@biemster how sure are we that this section is even needed?

biemster, 2025-05-05:
> [@biemster](https://github.com/biemster) how sure are we that this section is even needed? **EDIT:** I guess it's needed to compile with vendor libs.

Yes exactly, I want to RE the vendor libs so currently that's the use case, future users might want to do this because it will take quite a while to implement a full BLE stack.

commit 542c3bdf (2025-08-17): merge ch5[789]xhw.h headers into ch5xxhw.h, closes #594
closed 2025-08-17 by cnlohr