cnlohr/ch32fun #597: ch32v317 minichlink Chip type not supported

netmonk, 2025-05-13:
`[plonky@fedora minichlink]$ ./minichlink -i
Found WCH Link
WCH Programmer is LinkE version 2.16
Full Chip Type Reply: [8] 82-0d-05-86-31-73-b5-68-03
Detected: CH32V317
Setup success
Flash Storage: 288 kB
Part UUID    : 72-2a-da-98-89-6d-3c-ab
PFlags       : e3-39-e3-39
Part Type (B): 31-73-b5-68
Chip type not supported. Aborting...
Could not setup interface.`

Using wch linke: 

- swdio to swdio
- swcllk to swclk 
- 3.3v to 3.3v
- gnd to gnd

The dev board is freshly received ch32v317W-EVT-R0, i dont know if it's supported by minichlink

netmonk, 2025-05-14:
After running `minichlink -u` with some difficulties, it seems the issue is solved : 

```
[plonky@fedora minichlink]$ ./minichlink -u
Found WCH Link
Entering Unbrick Mode
Connection starting
DMStatus After Halt: /0/00000c82
Chip Type: 317
Unknown chip type.  Report as bug with picture of chip.
Vendored: 3173b568
marchid : dc68d886
HARTINFO: 00312380
Chip Type: 0
Chip Type: 317
Unknown chip type.  Report as bug with picture of chip.
Vendored: 3173b568
marchid : dc68d886
HARTINFO: 00312380
Chip Type: 317
Unknown chip type.  Report as bug with picture of chip.
Vendored: 3173b568
marchid : dc68d886
HARTINFO: 00312380
Chip Type: 317
Unknown chip type.  Report as bug with picture of chip.
Vendored: 3173b568
marchid : dc68d886
HARTINFO: 00312380
Chip Type: 317
Unknown chip type.  Report as bug with picture of chip.
Vendored: 3173b568
marchid : dc68d886
HARTINFO: 00312380
Chip Type: 317
Unknown chip type.  Report as bug with picture of chip.
Vendored: 3173b568
marchid : dc68d886
HARTINFO: 00312380
Chip Type: 317
Unknown chip type.  Report as bug with picture of chip.
Vendored: 3173b568
marchid : dc68d886
HARTINFO: 00312380
Chip Type: 317
Unknown chip type.  Report as bug with picture of chip.
Vendored: 3173b568
marchid : dc68d886
HARTINFO: 00312380
Chip Type: 317
Unknown chip type.  Report as bug with picture of chip.
Vendored: 3173b568
marchid : dc68d886
HARTINFO: 00312380
Chip Type: 317
Unknown chip type.  Report as bug with picture of chip.
Vendored: 3173b568
marchid : dc68d886
HARTINFO: 00312380
Chip Type: 317
Unknown chip type.  Report as bug with picture of chip.
Vendored: 3173b568
marchid : dc68d886
HARTINFO: 00312380
Chip Type: 317
Unknown chip type.  Report as bug with picture of chip.
Vendored: 3173b568
marchid : dc68d886
HARTINFO: 00312380
Chip Type: 317
Unknown chip type.  Report as bug with picture of chip.
Vendored: 3173b568
marchid : dc68d886
HARTINFO: 00312380
Chip Type: 317
Unknown chip type.  Report as bug with picture of chip.
Vendored: 3173b568
marchid : dc68d886
HARTINFO: 00312380
Chip Type: 317
Unknown chip type.  Report as bug with picture of chip.
Vendored: 3173b568
marchid : dc68d886
HARTINFO: 00312380
Chip Type: 317
Unknown chip type.  Report as bug with picture of chip.
Vendored: 3173b568
marchid : dc68d886
HARTINFO: 00312380
Chip Type: 317
Unknown chip type.  Report as bug with picture of chip.
Vendored: 3173b568
marchid : dc68d886
HARTINFO: 00312380
Chip Type: 317
Unknown chip type.  Report as bug with picture of chip.
Vendored: 3173b568
marchid : dc68d886
HARTINFO: 00312380
Whole-chip erase
Chip Type: 317
Unknown chip type.  Report as bug with picture of chip.
Vendored: 3173b568
marchid : dc68d886
HARTINFO: 00312380
[plonky@fedora minichlink]$ ./minichlink -i
Found WCH Link
WCH Programmer is LinkE version 2.15
link error, nothing connected to linker (4 = [81 55 01 02]).  Trying to put processor in hold and retrying.
link error, nothing connected to linker (4 = [81 55 01 02]).  Trying to put processor in hold and retrying.
Already Connected
Full Chip Type Reply: [4] 81-55-01-02-11-00-00-00-01
Unknown chip fallback
No detected chip.
Fault on op (DMABSTRACTS = 08000402) (100) (Processor not halted.) DMSTATUS: 000c0c82
Retrying
Setup success
Interface Setup
Chip Type: 317
Unknown chip type.  Report as bug with picture of chip.
Vendored: 3173b568
marchid : dc68d886
HARTINFO: 00312380
USER/RDPR  : c03f/5aa5
DATA1/DATA0: ff00/ff00
WRPR1/WRPR0: 00ff/00ff
WRPR3/WRPR2: 00ff/00ff
Flash Size: 288 kB
R32_ESIG_UNIID1: 722ada98
R32_ESIG_UNIID2: 896d3cab
R32_ESIG_UNIID3: e339e339
```

cnlohr, 2025-05-15:
How did you get it working with the 317?  Is there a PR?

netmonk, 2025-05-16:
Well i was suggested to proceed to a full erase of the chip and retry. 

Which was done with the `minichlink -u` at the begining. 

Once done, `minichlink -i` was able to proceed and didnt complain anymore. 

May be im wrong but i didnt add any code, therefore no PR for this issue.

cnlohr, 2025-05-16:
Thanks for clarification.