cnlohr/ch32fun #600: Getting CH572D to work

duddie, 2025-05-16:
Hi,

Has anyone succesfully got CH572 to work? I have two boards with CH572D. One is my own custom made and one original reference from WCH. Both boards do not work with minichlink. My V003 works fantastic and I have built whole environment around that so more or less I know what I am doing.

I get same result from both boards. Yet, the chip works in the USB mode as with shorted boot it reports itself as USB device.

(running minichlink under MacOS if this is important information)

If anyone has any idea I would be really happy to hear from you. Trying to build project quickly but got thsi obstacle.

**Here is log from CH572D:**
```

Found WCH Link
WCH Programmer is CH32V307 version 2.15
Already Connected
Full Chip Type Reply: [4] 81-55-01-01-11-00-00-00-01
Unknown chip fallback
No detected chip.
Fault on op (DMABSTRACTS = ffffffff) (-1) (Other Error) DMSTATUS: ffffffff
Retrying
Fault on op (DMABSTRACTS = ffffffff) (-1) (Other Error) DMSTATUS: ffffffff
Retrying
Fault on op (DMABSTRACTS = ffffffff) (-1) (Other Error) DMSTATUS: ffffffff
Retrying
Fault on op (DMABSTRACTS = ffffffff) (-1) (Other Error) DMSTATUS: ffffffff
Retrying
Fault on op (DMABSTRACTS = ffffffff) (-1) (Other Error) DMSTATUS: ffffffff
Retrying
Fault on op (DMABSTRACTS = ffffffff) (-1) (Other Error) DMSTATUS: ffffffff
Retrying
Fault on op (DMABSTRACTS = ffffffff) (-1) (Other Error) DMSTATUS: ffffffff
Retrying
Fault on op (DMABSTRACTS = ffffffff) (-1) (Other Error) DMSTATUS: ffffffff
Retrying
Fault on op (DMABSTRACTS = ffffffff) (-1) (Other Error) DMSTATUS: ffffffff
Retrying
Fault on op (DMABSTRACTS = ffffffff) (-1) (Other Error) DMSTATUS: ffffffff
Retrying
Fault on op (DMABSTRACTS = ffffffff) (-1) (Other Error) DMSTATUS: ffffffff
Retrying
Fault on setup -9
Could not setup interface.
```

**Here is log from working CH32V003**

```
Found WCH Link
WCH Programmer is CH32V307 version 2.15
Full Chip Type Reply: [8] 82-0d-05-09-00-30-05-00-03
Detected: CH32V003
Setup success
Flash Storage: 16 kB
Part UUID    : f7-5b-ab-cd-5f-7b-bc-53
PFlags       : ff-ff-ff-ff
Part Type (B): 00-30-05-00
Read protection: disabled
Interface Setup
USER/RDPR  : 00ff/5aa5
DATA1/DATA0: 00ff/00ff
WRPR1/WRPR0: 00ff/00ff
WRPR3/WRPR2: 00ff/00ff
Flash Size: 16 kB
R32_ESIG_UNIID1: f75babcd
R32_ESIG_UNIID2: 5f7bbc53
R32_ESIG_UNIID3: ffffffff
```

**Here is the CH572D chipset pic.** 

[image]

monte-monte, 2025-05-16:
Support for CH5xx is being tested in this git https://github.com/monte-monte/ch32v003fun/
Once everything is confirmed to be working it will get merged into main.

duddie, 2025-05-16:
@monte-monte 

Thank you so much for your information. I have upgraded to 2.16, used your version of minichlink but stillc cannot get information on CH572D. Verified that CH32V003 still works...

Current ouput from your version:

```

Found WCH Link
WCH Programmer is CH32V307 version 2.16
Already Connected
Unknown chip type.  Report as bug with picture of chip.
marchid : ffffffff
HARTINFO: ffffffff
[1]    10847 segmentation fault  ./minichlink -i

```

Let's ignore segmentation fault now (it's on MacOS) as it does not happen with CH32V003. 

Is there anything I can help from my side (I just am unsure where I could start) to help sort out this issue?

biemster, 2025-05-16:
> Is there anything I can help from my side (I just am unsure where I could start) to help sort out this issue?

Awesome to have a tester with a 572! Your programmer reports as `CH32V307` instead of `LinkE`, what exactly do you have?

It also seems the programmer is `Already Connected`, probably when you ran `make` it also fired the `minichlink` from main, which threw the error and got your whole setup in a connected state. Could you try with monte's minichlink on a freshly booted 572?

duddie, 2025-05-18:
Thank you all!

I had a lot of progress. First of all, my reference board had bad soldering for NRST (PA8) chip. After resoldering chip I have reference board fully working. So it was in first place hardware issue. The other board seem to have bad soldering problem too but I have not yet got to the time to do so.

I will work on this today and give further information.

duddie, 2025-05-18:
OK. This all was on hardware side... the other board (that I have designed) I forgot to ground the center pad... 🤪

Kids... don't forget to ground center pad... or you will have to drill hand made via 🥇 

Now with grounded chip it seems to work a little bit better 🙃

Thanks everyone for pointing me to monte-monte's toolchain. This is very helpful. If I find any problems I will report them quickly and hopefuly with solutions!

[image]

cnlohr, 2025-05-20:
Should this issue be closed?

leonheller, 2025-07-03:
I don't have any problems using the nanoCH57x from MuseLab with the MounRiver Studio IDE on my Windows laptop.

biemster, 2025-07-03:
> I don't have any problems using the WCH board and nanoCH57x from MuseLab with the MounRiver Studio IDE on my Windows laptop.

There are 2 things I like in this sentence, 2 I'm indifferent on and 2 I oppose with every fiber of my being. Quite a rollercoaster! It's good to know that WCH's own tools finally caught up with the 570, and support in `ch32fun` is imminent due to #630 👍 so I agree this can probably be closed.