cnlohr/ch32fun #603: Not an issue, but a ch32v003 web flash tool

olell, 2025-05-23:
Hey there, dunno if an Issue is the right place for this, but:

I just wanted to leave this here: I've created a little LED soldering kit / badge for my local hackspace with the CH32V003 and the rv003usb bootloader and wanted to make it as easy as possible to upgrade the firmware of this badge. So I've spent a couple of hours porting parts of the b003boot flash code of minichlink to TypeScript and building a flash tool using WebHID. I thought this is maybe useful for someone :)

The TypeScript library (I will release a first version as an npm package soon): https://github.com/olell/CH32WebFlash
And a simple web flashing tool build with React: https://github.com/olell/CH32WebFlashTool

You can try it out here: https://olell.github.io/CH32WebFlashTool/

This is all a bit hacky at the moment and definitively a prototype, but contributions and feedback are very welcome!

BogdanTheGeek, 2025-05-23:
@olell there is no easy way to say this, but I really wish you had joined the discord sooner. @monte-monte already wrote a tool that does the same thing here: https://subjectiverealitylabs.com/WeblinkUSB/

olell, 2025-05-27:
Yikes, but thank you for letting me know! I've read about the discord channel, but didn't really understand where to ping cnlohr for an invite link. If someone could invite me that would be really great!

monte-monte, 2025-05-28:
@olell is there any way to contact you in private?

closed 2025-05-28 by olell