cnlohr/ch32fun #624: malloc / free isn't implemented

jjflash65, 2025-06-22:
I have a few programs that I'm using with STM32 and I want to port them to CH32V003. These programs are using malloc and free from stdlib.h.

CH32FUN hasn't definied these function. Is there a chance to get them work ?

Nice greetings,
Ralph from germany

cnlohr, 2025-06-23:
You can also use various strip-down malloc/free implementations.  At least my preference is for ch32fun not to dictate what you should do here, because there are so many competing goals.  Some people may want heavier malloc/free's that avoid fragmentation better, others may have uniform-sized-objects, etc...  It's just hard for us to make a decision that catches everyone's needs.

cnlohr, 2025-06-23:
Should we make a statement about malloc/free on the readme?

> ## Malloc/free unimplemented in 003fun
> 
> Because 003fun services a wide variety of chips, it's hard to make decisions about what implementation of malloc/free should be used, comparing chips like the ch32v003 with 2kB of RAM, to the CH585 with 128kB of RAM, the decisions about what implementation to use will vary wildly. We've decided it's best for us not to decide for you.  You can choose to use none at all and just allocate statically, or use lists of fixed sizes, or use a heavier free/malloc.  Some malloc implementations you may want to consider are [tinyalloc](https://github.com/thi-ng/tinyalloc), [libmemory](https://github.com/embeddedartistry/libmemory), or other extremely lightweight but less common libraries like [mameMalloc](https://github.com/ar90n/mameMalloc),

cnlohr, 2025-06-23:
We can re-open discussion about this any time!! I am not fixed on this decision.  It's just to avoid letting this slip through the cracks, I am going to, for the time being say that statement.