cnlohr/ch32fun #660: blink does not work anymore with ardulink

zoobab, 2025-08-10:
Hi,

I am trying to flash a sop16 ch32v003 on a breakout board with an ardulink, but the blink example give this error (version 3abf32ba0bb1eb97ca4222cd05fa81cc3dd06369 of 8 aug):

```
../../ch32fun//../minichlink/minichlink -C ardulink -c /dev/ttyACM0 -w blink.bin flash -b
Opening serial port /dev/ttyACM0 at 115200 baud.
Ardulink: synced.
Interface Setup
Unknown chip type.  Report as bug with picture of chip.
marchid : ffffffff
HARTINFO: ffffffff
make: *** [../../ch32fun/ch32fun.mk:351 : cv_flash] Erreur 254
```

Now I tried my old personal fork of some months ago (see zoobab/ch32v003fun), and it works fine:

```
ch32v003fun/examples/blink$ ../../ch32v003fun/../minichlink/minichlink -c /dev/ttyACM0 -w blink.bin flash -b
Opening serial port /dev/ttyACM0 at 115200 baud.
Ardulink: synced.
Found Ardulink Programmer
Interface Setup
Image written.
```

zoobab, 2025-08-10:
I checked out 7e6e46e9d1e4b507f6d4e310b8e5e2497a02e53d , it still works:

```
../../ch32fun//../minichlink/minichlink -w blink.bin flash -C ardulink -c /dev/ttyACM0 -b
Opening serial port /dev/ttyACM0 at 115200 baud.
Ardulink: synced.
Interface Setup
Writing image
Chip Type: 071
Autodetected a SWDIO chip (Enum: 09 from 0713bb91)

Image written.
```

Let me try to pin point the culprit commit...

cnlohr, 2025-08-11:
@monte-monte did you want to bring a discussion here so others can find if they search?

cnlohr, 2025-08-12:
Wait, @monte-monte sorry, I just realized @BlueSyncLine is the one who contributed that, so they should be the first line of defense to fixing things up if possible.

closed 2025-08-13 by cnlohr