cnlohr/ch32fun #714: This is not an issue – just sharing a new portable IDE

NoNamedCat, 2025-09-30:
Hi everyone! 

I just wanted to share something I’ve been working on – a small, lightweight, and portable IDE called u-IDE.

Right now, it only includes CH32FUN as a supported platform (since that’s the environment I originally built it for). However, one of the nice things about this project is that it’s super easy to add new platforms — you just drop in a platform.json file with the necessary definitions.

The goal is to keep it simple, extensible, and easy to carry around ( PORTABLE! ). 

You can check it out here: [https://github.com/NoNamedCat/u-IDE](url)

[image]

I’d love to hear any thoughts, ideas, or feedback. Thanks for letting me share!

TommyMurphyTM1234, 2025-09-30:
> You can check it out here: [https://github.com/NoNamedCat/u-IDE](url)

The link doesn't work. I presume that it should be...

- https://github.com/NoNamedCat/u-IDE

NoNamedCat, 2025-10-03:
[image]

This is the examples menu.

cnlohr, 2025-10-08:
Would love for you to join the Discord to discuss more, please reach out to me at `cnlohr` on Discord for an invite.